pcdshub/pytmc #23: NameError: name 'pytmc' is not defined

mdavidsaver, 2019-02-17:
I updated to 2d1e6ce6125cb6e7ff230bf64e67e8411867edf4 and attempted to use the new `pytmc` executable.

```sh
$ pytmc './TwinCAT Project1/TwinCAT Project1/Untitled1/Untitled1.tmc' test
Traceback (most recent call last):
  File "/home/user/miniconda/envs/pytmc-env/bin/pytmc", line 11, in <module>
    load_entry_point('pytmc', 'console_scripts', 'pytmc')()
  File "/home/user/pytmc/pytmc/bin/makerecord.py", line 29, in main
    tmc_obj = pytmc.TmcFile(tmc_file)
NameError: name 'pytmc' is not defined
```

mdavidsaver, 2019-02-17:
I should note that I updated from 6bd1c9b4f8722266bf0b04aa6f5b045e4aa807da after seeing an error when attempting to process a non-trivial .tmc file https://github.com/craigmcchesney/XTES-SXR-vacuum-PLC-prototype/blob/dde825171530501fbc36c4ca3d991812e192c859/XtesSxrPlcProto/XtesSxrPlc/XtesSxrPlc.tmc

```sh
$ makerecord './XtesSxrPlcProto/XtesSxrPlc/XtesSxrPlc.tmc' XtesSxrPlc.db
Traceback (most recent call last):
  File "/home/user/miniconda/envs/pytmc-env/bin/makerecord", line 7, in <module>
    exec(compile(f.read(), __file__, 'exec'))
  File "/home/user/pytmc/bin/makerecord", line 42, in <module>
    main()
  File "/home/user/pytmc/bin/makerecord", line 32, in main
    tmc_obj.isolate_chains()
  File "/home/user/pytmc/pytmc/xml_collector.py", line 319, in isolate_chains
    new_singular_chains = non_singular.build_singular_chains()
  File "/home/user/pytmc/pytmc/xml_collector.py", line 495, in build_singular_chains
    config.pragma.fix_to_config_name(select_name[0])
AttributeError: 'NoneType' object has no attribute 'fix_to_config_name'
```

mdavidsaver, 2019-02-20:
@n-wbrown Will you have time to look into this soon?  (Today?)

n-wbrown, 2019-02-20:
I'm currently unable to recreate the issue you presented in your first comment (commit  [2d1e6ce](https://github.com/slaclab/pytmc/commit/2d1e6ce6125cb6e7ff230bf64e67e8411867edf4)). Did you make sure to uninstall the old package and install the new one?

If you tag my name on the first comment it will help me see your issues and treat them much faster.

mdavidsaver, 2019-02-20:
I am easily able to replicate in a freshly installed VM using a process taken from your .travis.yml

```sh
git clone https://github.com/slaclab/pytmc
cd pytmc
wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O ~/miniconda.sh
bash ~/miniconda.sh -b -p $HOME/miniconda
export PATH="$HOME/miniconda/bin:$PATH"
conda install conda-build anaconda-client pip
conda update -q conda
conda create -n pytmc-env pip python
source activate pytmc-env
pip install -r requirements.txt
pip install -e .

cd ..
wget https://github.com/craigmcchesney/XTES-SXR-vacuum-PLC-prototype/raw/dde825171530501fbc36c4ca3d991812e192c859/XtesSxrPlcProto/XtesSxrPlc/XtesSxrPlc.tmc
pytmc XtesSxrPlc.tmc output.db
Traceback (most recent call last):
  File "/home/user/miniconda/envs/pytmc-env/bin/pytmc", line 11, in <module>
    load_entry_point('pytmc', 'console_scripts', 'pytmc')()
  File "/home/user/pytmc/pytmc/bin/makerecord.py", line 29, in main
    tmc_obj = pytmc.TmcFile(tmc_file)
NameError: name 'pytmc' is not defined
```

mdavidsaver, 2019-02-20:
```sh
$ grep -w pytmc /home/user/pytmc/pytmc/bin/makerecord.py
    "pytmc" is a command line utility for generating epics records files from
    tmc_obj = pytmc.TmcFile(tmc_file)
```

n-wbrown, 2019-02-20:
I've fixed the first issue [here](https://github.com/n-wbrown/pytmc/tree/fixes-2-20).

I'm investigating the second now.

n-wbrown, 2019-02-20:
@mdavidsaver. The primary bug we've been experiencing has been fixed in PR #24, with this [commit](https://github.com/slaclab/pytmc/pull/24/commits/a0b7c8623d06df09a3ff48c3c9c3048b41cf2161).  I'm not merging the PR in yet because there still remains a small bug where too many colons are drawn in the PV name. 

The PR in its current state should be sufficient for testing purposes. Despite being ugly, the extra colons issue is mostly aesthetic. Functional IOCs should be possible but the PV names will be odd. I'll be working to fix this over the next few days and once this is in, we should be able to regenerate the DB files.

mdavidsaver, 2019-02-24:
@n-wbrown I tried running this script again, and it now succeeds.  However, the result has several errors which prevent it from being loaded.  Did you test this?

```
dbLoadRecords("XtesSxrPlc.db","P=ADS_IOC:OCTET:,PORT=ADS_1")
Can't set "MR2K3_GPI_1::PRESS.DTYP" to "asynFloat32" No digits to convert
Error at or before ")" in path "."  file "XtesSxrPlc.db" line 15
Can't set "MR2K3_GPI_1::VAC_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR2K3_GPI_1::PRO_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR2K3_GPI_1::SP_HYS.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "TV2K4_GPI_1::PRESS.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "TV2K4_GPI_1::VAC_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "TV2K4_GPI_1::PRO_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "TV2K4_GPI_1::SP_HYS.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR1K4_GPI_1::PRESS.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR1K4_GPI_1::VAC_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR1K4_GPI_1::PRO_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR1K4_GPI_1::SP_HYS.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR2K3_GCC_1::PRESS.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR2K3_GCC_1::VAC_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR2K3_GCC_1::PRO_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR2K3_GCC_1::SP_HYS.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "TV2K4_GCC_1::PRESS.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "TV2K4_GCC_1::VAC_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "TV2K4_GCC_1::PRO_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "TV2K4_GCC_1::SP_HYS.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR1K4_GCC_1::PRESS.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR1K4_GCC_1::VAC_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR1K4_GCC_1::PRO_SP.DTYP" to "asynFloat32" No digits to convert
ErrorCan't set "MR1K4_GCC_1::SP_HYS.DTYP" to "asynFloat32" No digits to convert
ErrorBad character ' ' in record name "TV2K4_VGC_1::OPN_SW field: ZNAM CLOSE field: ONAM OPEN io: io"
Can't create record "TV2K4_VGC_1::OPN_SW field: ZNAM CLOSE field: ONAM OPEN io: io" of type "bo"
ErrorError: syntax error
```

Record names can't contain spaces, and can't be this long.  "asynFloat32" is also not valid for an aiRecord, try "asynFloat64" (cf. ```dbd/devAsynFloat64.dbd``` from the asyn module).

```sh
$ grep record /tmp/XtesSxrPlc.db
...
record(bo,"TV2K4_VGC_1::OPN_SW field: ZNAM CLOSE field: ONAM OPEN io: io"){
record(bo,"TV2K4_VGC_1::ALM_RST"){
record(bo,"TV2K4_VGC_1::FORCE_OPN"){
record(bo,"TV2K4_VGC_1::OVRD_ON io: i"){
record(bi,"TV2K4_VGC_1::DP_OK"){
record(bo,"TV2K4_VGC_1::Ext_ILK_OK io: i"){
```

mdavidsaver, 2019-02-24:
At least one of the problems on display here seems to be incorrectly formatted annotations (missing line breaks) in the L2Si_Vacuum_Library.

```
    <DataType>
      <Name Namespace="L2Si_Vacuum_Library">ST_VGC</Name>
...
      <SubItem>
        <Name>pv_xOPN_SW</Name>
        <Type>BOOL</Type>
        <Comment> EPICS Controls </Comment>
        <BitSize>8</BitSize>
        <BitOffs>24</BitOffs>
        <Properties>
          <Property>
            <Name>pytmc</Name>
            <Value>pv: OPN_SW field: ZNAM CLOSE field: ONAM OPEN io: io </Value>
          </Property>
        </Properties>
      </SubItem>
      <SubItem>
        <Name>pv_xAlmRst</Name>
        <Type>BOOL</Type>
        <BitSize>8</BitSize>
        <BitOffs>32</BitOffs>
        <Properties>
          <Property>
            <Name>pytmc</Name>
            <Value>
    pv: ALM_RST
        io: o
    </Value>
          </Property>
        </Properties>
      </SubItem>
```

@craigmcchesney

n-wbrown, 2019-03-07:
Pytmc has been patched to support semicolons as delimiters (See #25). Now working on adding semicolons to the libraries.

mdavidsaver, 2019-03-14:
With 619562e9dde4731716bd797c735219c8c6bb94c1

Trying again with https://github.com/craigmcchesney/XTES-SXR-vacuum-PLC-prototype/commit/ebcb0d355e9bfae8835a790c94ab54b61616547a/XtesSxrPlcProto/XtesSxrPlc/XtesSxrPlc.tmc

I get:

```
$ pytmc XtesSxrPlc.tmc output.db 
Traceback (most recent call last):
  File "/home/user/miniconda/envs/pytmc-env/bin/pytmc", line 11, in <module>
    load_entry_point('pytmc', 'console_scripts', 'pytmc')()
  File "/home/user/pytmc/pytmc/bin/makerecord.py", line 41, in main
    tmc_obj = pytmc.TmcFile(tmc_file)
  File "/home/user/pytmc/pytmc/xml_collector.py", line 100, in __init__
    self.isolate_all()
  File "/home/user/pytmc/pytmc/xml_collector.py", line 207, in isolate_all
    self.isolate_DataTypes()
  File "/home/user/pytmc/pytmc/xml_collector.py", line 148, in isolate_DataTypes
    self.isolate_SubItems(data.name)
  File "/home/user/pytmc/pytmc/xml_collector.py", line 167, in isolate_SubItems
    parent=parent_obj
  File "/home/user/pytmc/pytmc/xml_obj.py", line 1036, in __init__
    super().__init__(element, base, suffixes)
  File "/home/user/pytmc/pytmc/xml_obj.py", line 560, in __init__
    self.pragma = Configuration(self.raw_config)
  File "/home/user/pytmc/pytmc/xml_obj.py", line 34, in __init__
    self.config = self._formatted_config_lines()
  File "/home/user/pytmc/pytmc/xml_obj.py", line 143, in _formatted_config_lines
    config_lines = self._config_lines()
  File "/home/user/pytmc/pytmc/xml_obj.py", line 92, in _config_lines
    line_parser.search(m).groupdict() for m in result_no_delims
  File "/home/user/pytmc/pytmc/xml_obj.py", line 92, in <listcomp>
    line_parser.search(m).groupdict() for m in result_no_delims
AttributeError: 'NoneType' object has no attribute 'groupdict'
```

klauer, 2019-03-18:
Can we close this and move to more specific issues?

mdavidsaver, 2019-03-18:
I assume you mean #27?  Fine by me as long as the interested parties are kept aware.  @craigmcchesney

closed 2019-03-18 by mdavidsaver