pcdshub/pytmc #27: Parsing failure in tmc file (None has no attribute 'groupdict')

klauer, 2019-03-18:
_Originally posted by @mdavidsaver in https://github.com/slaclab/pytmc/issues/23#issuecomment-473076602_

With 619562e9dde4731716bd797c735219c8c6bb94c1

Trying again with https://github.com/craigmcchesney/XTES-SXR-vacuum-PLC-prototype/commit/ebcb0d355e9bfae8835a790c94ab54b61616547a/XtesSxrPlcProto/XtesSxrPlc/XtesSxrPlc.tmc

I get:

```
$ pytmc XtesSxrPlc.tmc output.db 
Traceback (most recent call last):
  File "/home/user/miniconda/envs/pytmc-env/bin/pytmc", line 11, in <module>
    load_entry_point('pytmc', 'console_scripts', 'pytmc')()
  File "/home/user/pytmc/pytmc/bin/makerecord.py", line 41, in main
    tmc_obj = pytmc.TmcFile(tmc_file)
  File "/home/user/pytmc/pytmc/xml_collector.py", line 100, in __init__
    self.isolate_all()
  File "/home/user/pytmc/pytmc/xml_collector.py", line 207, in isolate_all
    self.isolate_DataTypes()
  File "/home/user/pytmc/pytmc/xml_collector.py", line 148, in isolate_DataTypes
    self.isolate_SubItems(data.name)
  File "/home/user/pytmc/pytmc/xml_collector.py", line 167, in isolate_SubItems
    parent=parent_obj
  File "/home/user/pytmc/pytmc/xml_obj.py", line 1036, in __init__
    super().__init__(element, base, suffixes)
  File "/home/user/pytmc/pytmc/xml_obj.py", line 560, in __init__
    self.pragma = Configuration(self.raw_config)
  File "/home/user/pytmc/pytmc/xml_obj.py", line 34, in __init__
    self.config = self._formatted_config_lines()
  File "/home/user/pytmc/pytmc/xml_obj.py", line 143, in _formatted_config_lines
    config_lines = self._config_lines()
  File "/home/user/pytmc/pytmc/xml_obj.py", line 92, in _config_lines
    line_parser.search(m).groupdict() for m in result_no_delims
  File "/home/user/pytmc/pytmc/xml_obj.py", line 92, in <listcomp>
    line_parser.search(m).groupdict() for m in result_no_delims
AttributeError: 'NoneType' object has no attribute 'groupdict'
```

klauer, 2019-03-19:
I can't reproduce the original exception with master (v1.1.2). However, `pytmc` erroneously outputs records such as:

```c
record(bo,"TV2K4_VGC_1:OPN_SW field: ZNAM CLOSE field: ONAM OPEN io: io"){
  info(asyn:READBACK,"1")
  field(PINI, "1")
  field(TSE, -2)
  field(DTYP, "asynInt32")
  field(OUT, "@asyn($(PORT),0,1)ADSPORT=851/GVL_DEVICES.TV2K4_VGC_1.iq_stValve.pv_xOPN_SW?")
  field(SCAN, "Passive")
  field(ONAM, One)
  field(ZNAM, Zero)
}
```

Note, specifically, the record name: `"TV2K4_VGC_1:OPN_SW field: ZNAM CLOSE field: ONAM OPEN io: io"`

klauer, 2019-03-19:
OK, it appears that the above is due to improperly formatted tags in the .tmc file itself:

Problematic version: https://github.com/craigmcchesney/XTES-SXR-vacuum-PLC-prototype/blob/ebcb0d355e9bfae8835a790c94ab54b61616547a/XtesSxrPlcProto/XtesSxrPlc/XtesSxrPlc.tmc#L75

The recent master corrects this:
https://github.com/craigmcchesney/XTES-SXR-vacuum-PLC-prototype/blob/c275fe633f27adfab86f31c001725f9b06342b52/XtesSxrPlcProto/XtesSxrPlc/XtesSxrPlc.tmc#L81

Resulting in:
```c
record(bo,"TV2K4_VGC_1:OPN_SW"){
  info(asyn:READBACK,"1")
  field(ONAM, OPEN)
  field(PINI, "1")
  field(TSE, -2)
  field(DTYP, "asynInt32")
  field(OUT, "@asyn($(PORT),0,1)ADSPORT=851/GVL_DEVICES.TV2K4_VGC_1.iq_stValve.pv_xOPN_SW?")
  field(SCAN, "Passive")
  field(ZNAM, Zero)
}
```

It seems like the parsing could be a bit smarter in detecting section breaks (either raising an error or parsing it correctly), but that's probably a separate issue.

OK to close this @craigmcchesney @n-wbrown @mdavidsaver?

n-wbrown, 2019-03-19:
I think this is good to close. The fix has come in two parts: 
1. Pytmc can accept a wider variety of delimiters on the configuration line e.g. new-lines, semicolons, and the end of the pragma string (useful for single line pragmas)
2. The vacuum library is being updated with proper delimiters. (Still in progress) 

Unless this error has occurred in a new or exciting way, It's safe to close this issue. 

I like the suggestion of alerting the user to these issues or automatically correcting them.

closed 2019-03-19 by n-wbrown

mdavidsaver, 2019-03-20:
With 6c8a5305aa5efbd08132e48bf32db0e3bda1fb92 I can now generate a database which loads without error.