pcdshub/pytmc #29: Idea: EPICS database and tmc file versioning synchronization

klauer, 2019-03-19:
Consider generating a read-only record with a unique identifier of the `.tmc` file + version:
* If in a *clean* git repository, `git describe --tags`
* (All/part of) the `.tmc` sha-sum or some other hash
* Something else?

n-wbrown, 2019-03-20:
Additional information: 
 - Creation date
 - Author name?
 - If we end up in a situation where we have multiple EPICS ADS Module versions we could specify the version this file is intended to work with

klauer, 2023-11-08:
We work around this by way of generating the version records inside the IOC startup script. It makes more sense than baking it into the database file, I think.

Closing as "fixed a different way"

closed 2023-11-08 by klauer