pcdshub/pytmc #36: Generated .db contains syntax and logic errors

mdavidsaver, 2019-03-31:
Using a53733271477f0880e58529e19246f290c32ae88 to process https://github.com/craigmcchesney/XTES-SXR-vacuum-PLC-prototype/blob/c275fe633f27adfab86f31c001725f9b06342b52/XtesSxrPlcProto/XtesSxrPlc/XtesSxrPlc.tmc results in a .db file which fails to load.

```
dbLoadRecords("XtesSxrPlc.db","P=ADS_IOC:OCTET:,PORT=ADS_1")
Can't set "MR2K3_GPI_1:PRESS.DTYP" to "asynFloat32" No digits to convert
Error at or before ")" in path "."  file "XtesSxrPlc.db" line 15
Error: syntax error
```

mdavidsaver, 2019-03-31:
Running my [dbdlint](https://github.com/epicsdeb/pypdb.git) gives:

```sh

mdavidsaver, 2019-03-31:
Running my [dbdlint](https://github.com/epicsdeb/pypdb.git) gives:

```sh
$ dbdlint -Wno-quoted -F ../m-epics-twincat-ads/dbd/adsExApp.dbd XtesSxrPlc.db 
ERROR   XtesSxrPlc.db:961 empty - Empty string not allowed
ERROR   XtesSxrPlc.db:961 bad-rtyp - TV2K4_VGC_1:ErrMsg has unknown record type ''
ERROR   XtesSxrPlc.db:1179 empty - Empty string not allowed
ERROR   XtesSxrPlc.db:1179 bad-rtyp - MR1K4_VGC_1:ErrMsg has unknown record type ''
ERROR   XtesSxrPlc.db:15 bad-dtyp - record type 'ai' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:104 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:114 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:124 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:143 bad-dtyp - record type 'ai' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:232 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:242 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:252 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:271 bad-dtyp - record type 'ai' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:360 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:370 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:380 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:399 bad-dtyp - record type 'ai' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:488 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:498 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:508 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:527 bad-dtyp - record type 'ai' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:616 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:626 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:636 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:655 bad-dtyp - record type 'ai' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:744 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:754 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:764 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:917 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:926 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:935 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:969 bad-field - recordtype 'ai' has not field 'THST'
ERROR   XtesSxrPlc.db:1135 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:1144 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:1153 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:1187 bad-field - recordtype 'ai' has not field 'THST'
ERROR   XtesSxrPlc.db:1230 bad-dtyp - record type 'ao' has no DTYP 'asynFloat32'
ERROR   XtesSxrPlc.db:1260 bad-dtyp - record type 'ai' has no DTYP 'asynFloat32'
```

n-wbrown, 2019-06-26:
Closed by #49

closed 2019-06-26 by n-wbrown