pcdshub/pytmc #41: Archive and Autosave Designation

teddyrendahl, 2019-04-15:
I think it iscrucial to allow a user to request that a specific PV created by a `pytmc` database file is archived or autosaved. Ideally this is just an additional pragma, but putting it somewhere where we can make sure that the built IOC respects these takes some thought.

klauer, 2019-07-22:
For autosave:
* `info` nodes should be used. Example from the [documentation](https://epics.anl.gov/bcda/synApps/autosave/autoSaveRestore_R5-6-1.html):
```
record(ao, "$(P)test1") {
  field(DTYP, "Soft Channel")
  info(autosaveFields, "PREC EGU DESC")
  info(autosaveFields_pass0, "VAL")
}
From this information, makeAutosaveFiles() will write the following two files:

info_settings.req
$(P)test1.PREC
$(P)test1.EGU
$(P)test1.DESC

info_positions.req
$(P)test1.VAL
This function can be called at any time after iocInit().
```
* Additional fields could be set on a per-record basis
* Likely all setpoint .VAL should be saved by default

klauer, 2019-11-25:
As #137 will tackle autosave, I'm going to split the archiver into a separate issue.

PR #137 (2019-12-06): ENH: add autosave support
Closes #41 

Defaults are specified in classes from `pytmc.record`. These can be overridden by using specific keys in the pragmas:

```
        To apply to either input or output records, pragma keys `autosave_pass0`
        or `autosave_pass1` can be used.

        To only apply to input records, pragma keys `autosave_input_pass0`
        `autosave_input_pass1` can be used.

        To only apply to output records, pragma keys `autosave_output_pass0`
        `autosave_output_pass1` can be used.
```

More info in the documentation commit (see diff).

I have not given  [... cut]
closed 2019-12-06 by klauer