pcdshub/pytmc #50: Anaconda build failing

n-wbrown, 2019-04-17:
The travis builds are breaking here:

https://travis-ci.org/slaclab/pytmc/jobs/520858300

```sh
$ if [[ $TRAVIS_PULL_REQUEST == false && $TRAVIS_REPO_SLUG == $OFFICIAL_REPO && $TRAVIS_BRANCH == 'master' ]]; then
      export ANACONDA_API_TOKEN=$CONDA_UPLOAD_TOKEN_DEV
      anaconda upload bld-dir/linux-64/*.tar.bz2
    fi
    
Using Anaconda API: https://api.anaconda.org
The action you are performing requires authentication, please sign in:
Using Anaconda API: https://api.anaconda.org
Username: 
No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
The build has been terminated
```

teddyrendahl, 2019-04-17:
✅ 
```
conda install pytmc -c pcds-dev --dry-run
Solving environment: done

The following NEW packages will be INSTALLED:

    pytmc:           1.1.2+59.g053a9a3-py36_1 pcds-dev   
```

n-wbrown, 2019-04-17:
Thanks @teddyrendahl!

closed 2019-04-17 by n-wbrown