pcdshub/pytmc #76: Commit Hooks for PLC Projects

teddyrendahl, 2019-04-30:
Twincat developers want to confirm that between commits they have not made changes to the PLC project that will adversely affect the corresponding EPICS IOC. The solution could be that when you push to a GitHub branch, we run `pytmc` and check that the generated EPICS database is valid via Travis.

This will require including `.tmc` files in Git version control.

slacAWallace, 2019-04-30:
We track TMC already. Maybe we should mark it as binary?

klauer, 2019-04-30:
Great to hear we're keeping track of tmc files in the git repos. It's XML, so text would make sense.

Workflow as I imagine it:
1. Build project, commit via git
2. Push to GitHub
3. Travis
    a. Builds the IOC, generating a database definition file (dbd)
    b. Runs pytmc, generating a database
    c. pytmc integrates with pypdb, linting the generated database file against the dbd
4. Travis passes/fails based on the results of the above

With a pass on step (4), PLC authors could have reasonable assurance the IOC would at least run.

klauer, 2019-08-15:
Partial solution for improving TMC diffs: https://gist.github.com/klauer/a1489e408042a95ed9e804e37e7832c2