pcdshub/pytmc #77: pyads Server for testing

teddyrendahl, 2019-04-30:
Stand-up a `pyads.TestServer` to use a mock "PLC" we can then load an EPICS database file and query via Channel Access whether the records respond accordingly.

slacAWallace, 2019-04-30:
Does this belong in the ADS module testing suite?

klauer, 2019-04-30:
@slacAWallace My 2c on it: since pytmc is the place we interpret tmc files and make EPICS IOCs, pytmc is probably also the right place for a pyads server that stands up a mock-PLC based on that same TMC file.

klauer, 2019-08-13:
I've experimented a bit with the pyads-provided server implementation, and it would need a fair amount of work for this to be feasible, sadly.

klauer, 2020-08-31:
Two paths forward since this issue was opened. In some scenarios one might be chosen over the other:
* (ADS layer) I have some hope for https://github.com/pcdshub/ads-async - but no time to work on it. It has a preliminary ads server that uses .tmc files to create a map of PLC memory + expose symbols over the ADS protocol. It preliminarily works with a real IOC talking to it (see the two open PRs)
* (IOC layer) ads-deploy can auto-create a mock caproto IOC from a TwinCAT project - with no PLC/ADS communication

closed 2020-08-31 by klauer