pcdshub/pytmc #95: Duplicated ai and longin records

domitto, 2019-07-31:
Hi, it looks like pytmc is generating duplicate records of type ai and longin.

This is the tmc file I've used: https://github.com/craigmcchesney/GmdPlc/blob/master/GmdPlc/GmdPlc/GmdPlc.tmc

I've also attached the generated db file.
[gmd.zip](https://github.com/slaclab/pytmc/files/3449317/gmd.zip)

klauer, 2019-07-31:
Hi @domitto, thanks for the report! (Our work has overlapped quite a bit today, hasn't it? :) )

It looks like the L2Si vacuum library pragmas need fixing. There are two items associated with `FB_MKS422` that create the PV suffix `PRESS`:

1. `i_PRESS_R`:
[image]

2. `rPRESS`:
[image]

(cc @ghalym @slacawallace)

domitto, 2019-07-31:
Hi @klauer,
Thanks for looking into that (man, you're in all the projects 😁).
It would be nice as a future feature to throw an error or warning saying that the pragmas are duplicated in the input file.

ghalym, 2019-07-31:
@klauer @domitto 

Thank you for catching that. I have fixed the duplicate pragma and also updated some pragma names. I pushed the changes to the pcdshub/L2SI-Vacuum-Lib.

(cc @slacAWallace)

klauer, 2019-07-31:
Great! Thanks for the quick fix @ghalym 😄 

Agreed, @domitto, I think pytmc can do better and warn/error out in this specific case. 
I think we can close this for now, but re-open with an issue stating just that.

A current workaround is that if you use the linter option `--dbd` (`pytmc db --dbd` in the current version) it will fail with an error saying that you're changing the record type from `ai` to `longin` or something along those lines.

closed 2019-07-31 by klauer