pcdshub/pytmc #132: TwinCAT does not currently allow '$' in a pragma

klauer, 2019-11-21:
With `$` in the pragma, the entire pragma is ignored and not included in the `.tmc`.

* pytmc pragmalint should catch this and warn
* An alternative to macro substitution strings should be proposed. Could we replace `%()` in values with `$()` for example? (cc @n-wbrown @ZLLentz)
* Should report this to Beckhoff as well

ZLLentz, 2019-11-21:
I don't have a preference for replacement symbols. Percent is fine

n-wbrown, 2019-11-21:
There's a possibility this could conflict with the way arrays of objects are formatted if we're using the `%` symbol to specify how the array's index will be formatted in the PV name. Either way it's probably good practice to allow us to change exactly which symbol we use for this at a later date. `#`, `@`, `&`. and `^` could be good replacements if we need them. Maybe we start with one of the other symbols so if Beckhoff hasn't patched the `$` rejection by the time we need it, we don't have to create a second interim convention.

klauer, 2019-11-21:
Good point, @n-wbrown, I think we probably should avoid `%` for your stated reason...

klauer, 2019-11-27:
#135 will close this, using `@` as the alternate character for `pv:` strings.

PR #135 (2019-11-27): ENH/FIX: Better linting, debug tool fixes, $ handling, ... and more!
Closes #134 
Closes #127 
Closes #131 
Closes #133 
Closes #132
closed 2019-11-27 by klauer