pcdshub/pytmc #139: Support the EPICS archiver appliance

klauer, 2019-11-25:
From Murali, we now know that:

```
Cron jobs are in place running processRequestFile.py, so just make sure your IOC build and
release process installs archive request files to $IOC_DATA/$IOC_NAME/archive/*.archive w/ syntax
MY:PV:NAME    period    [ type ]
period can be integer or floating point
type is optional, either monitor or scan, defaults to scan.
```

We can create our own `info` tags per-record and generate these files. Or `pytmc` could output them directly.

klauer, 2019-12-06:
Default will be `scan` `1s`, except for waveforms

klauer, 2019-12-06:
(*) Waveforms of larger than a threshold, that is

klauer, 2019-12-06:
`archive: {1s/1hz} {scan/monitor}`
Default: `archive 1s scan`

Examples:
`archive: 1s scan`
`archive: 1hz monitor`
`archive: 1s` (=scan)

n-wbrown, 2019-12-06:
Notes from discussion with @klauer 

Reuse the `bin.bin.process method for generating a list of records. 

Add archive file generation onto the `db` command line tool. When an output file is specified, the archive file will be generated by default.

PR #162 (2019-12-19): ENH: add preliminary archive file support
Closes #139

See the related issue + docs for further information.

- [x] Fix nested imports
- [x] Add better tests
- [x] Docs
closed 2019-12-19 by klauer