pcdshub/pytmc #146: Proposed syntax for controlling ADS poll rate

n-wbrown, 2019-12-05:
Proposed syntax for controlling ADS poll rate.

```
poll:
    init
    no[tify]
    1hz
    1s[ec]
```

klauer, 2019-12-05:
Agreed-upon `INP @asyn` options:
- `POLL_RATE=[hz]/` (polling) 
- `TS_MS=[ms]/` (notifications)

IOC startup will have configurable default polling rate

`poll: init` won't be an option, and I think we need to rethink the pragma syntax given the above

n-wbrown, 2019-12-06:
how about this:
```
poll:
    0.1s        # Notify, rate limited to 10hz
    poll=1hz    # Attach to the 1hz poll group
```
I'm not sure I like specifying T for the notify rate but if we specify frequency, we flip the convention that asyn is using. This does seem to be less than ideal for the intended use cases for both types of communication.

I'm also not happy about the double use of the `poll` keyword in this draft. Very open to better ideas.

klauer, 2019-12-06:
How about consistent handling of the time/frequency, but then just having 'notify' as a separate word?

`poll: 1hz notify` = `poll: 1s notify`
`poll: 1s` = `poll: 1hz`

Or:

`update: 1hz notify` = `update: 1s notify`
`update: 1s poll` = `update: 1hz poll` = `update: 1s` (default is 'poll')

klauer, 2019-12-06:
@n-wbrown and I discussed in person and like `update`.

If the pragma is unspecified, the default is assumed to be `update: 1s poll`.
We will not support the additional `sec` vs `s` or variants thereof.

PR #151 (2019-12-13): ENH: allow configuration of update rate (poll/notify) via pragma
The goal is to support specification of either polling/notification with the new [m-epics-twincat-ads bulk polling](https://github.com/pcdshub/twincat-ads/commit/a24bbd862614bb2ad51d2f648ee67bb46cedddba).

See the diff for docs regarding the `update` pragma key.

Closes #146 
Closes #153 

TODO: eventually should restructure, moving the independent pragma normalization functions out to `pytmc.parser` (or elsewhere) to avoid nesting the imports
closed 2019-12-13 by n-wbrown