pcdshub/pytmc #149: Conda Recipe Missing epics-pypdb Dependency

ZLLentz, 2019-12-06:
See title

klauer, 2019-12-07:
We're missing pypdb on conda entirely :(

ZLLentz, 2019-12-07:
:(

klauer, 2019-12-07:
Paging conda-forge expert @hhslepicka

hhslepicka, 2019-12-07:
I asked Michael to cut an official release of this package. When we have it I will add it to conda-forge.

ZLLentz, 2019-12-09:
Ok, so actually the `pypdb` dependency is listed in the conda recipe, but the package requires the dependency to explicitly be called `epics-pypdb` when run via the `pytmc` script that gets built in a python environment. This is a script you never see unless you build/install the package. I'm going to make a PR for testing that script.

hhslepicka, 2019-12-10:
A conda recipe was added to https://github.com/pcdshub/epics-pypdb so it can now be built locally.
Also, once this PR ( https://github.com/conda-forge/staged-recipes/pull/10408 ) is merged we will have it available via `conda-forge` as well.

ZLLentz, 2019-12-10:
Thank you! Should I rebuild locally now or did you do this too?

hhslepicka, 2019-12-10:
No... I did not build it locally... 😞

PR #156 (2019-12-10): BLD: rename pypdb to epics-pypdb in conda recipe
Just doing the easy part

closes #149
closed 2019-12-10 by klauer