pcdshub/pytmc #179: Summary, pragmalint does not locate independent IO files properly

n-wbrown, 2020-02-20:
Instead of seeking the .xti files in the `[project_name]/_Config/IO/*.xti` summary and pragmalint attempt to locate the files in `[project_name]/[project_name]/*.xti`. 

Creating symlinks provides a temporary hack/solution to resolve this issue.

klauer, 2020-02-20:
Do you have a sample project repo to add to our test suite?

n-wbrown, 2020-02-20:
https://github.com/n-wbrown/lcls-plc-kfe-gmd-vac/commit/4b4105124f5ca00aac6353d7b8c58e12ac4ba333

This is the commit prior to adding the symlinks.

I plan to merge this into master soon

klauer, 2020-02-20:
I see the what's happening, but I wonder what the environment configuration differences are coming in here. I'd like for pytmc to support any setting, but for now can you compare your settings against https://confluence.slac.stanford.edu/display/PCDS/TwinCAT+3+Git+Setup+and+Best+Practices ?

PR #180 (2020-02-26): FIX: config/io path loading
I don't have much faith in the correctness of this, but it appears to load all (valid) tsproj files I have on hand, including those in the test suite.

* Added https://github.com/n-wbrown/lcls-plc-kfe-gmd-vac/commit/4b4105124f5ca00aac6353d7b8c58e12ac4ba333 as a submodule, which should close #179
closed 2020-02-26 by n-wbrown