pcdshub/pytmc #181: Describe cause/origin of exceptions discovered when running bi.db.process

n-wbrown, 2020-02-22:
https://github.com/slaclab/pytmc/blob/master/pytmc/bin/db.py#L119

The issue with the record is well described: `        logger.error('Error creating record: %s', ex)` but the record for which creation was unsuccessful is not shown

n-wbrown, 2020-02-22:
Issue can be seen in the .tmc file here: https://github.com/n-wbrown/lcls-plc-kfe-gmd-vac/commit/8213c3ccba99193ac87bd6f2a3206bd2bb032beb

n-wbrown, 2020-02-22:
More specifically, this is the error being provided: `ERROR:pytmc.bin.db:Error creating record: 'EnumInfo' object has no attribute 'Enum'`

n-wbrown, 2020-02-24:
Note from conversation: expanded report on guilty chain is useful but should be relegated to a verbose option. It's way too much information for normal pytmc usage.

klauer, 2020-02-24:
As we discussed in person, let's definitely show the chain name here.

n-wbrown, 2020-02-24:
Started #184 to add more verbose (but not too verbose) errors.

klauer, 2020-02-26:
Merging #184, but will need to circle back to keep the exception traceback intact.