pcdshub/pytmc #187: Path loading still isn't quite right (independent project files)

klauer, 2020-02-26:
Leaving everything to be "independent" in TwinCAT, pytmc has no issue on any project I've found so far:

[image]

Changing these values or manually tweaking a file's status as an external ("independent project file") can cause the parser to fail to find the file:

[image]

At some point, we should generate all feasible variants of projects and ensure that pytmc can load them all.

slacAWallace, 2020-07-08:
I ran into this, or something similar today.

https://travis-ci.com/github/pcdshub/lcls-plc-kfe-arbiter/builds/174854952

klauer, 2020-11-03:
I think this did it: https://github.com/slaclab/pytmc/pull/228 (fingers crossed)

closed 2020-11-03 by klauer