pcdshub/pytmc #190: BOOL ONAM/ZNAM Defaults

ZLLentz, 2020-03-24:
Currently the defaults for `BinaryRecordPackage` are:  https://github.com/slaclab/pytmc/blob/7290ba67a91fca2970a4311cf3a165ab7c5cc696/pytmc/record.py#L471

This is unintuitive because the only TwinCAT type that maps to `BinaryRecordPackage` is `BOOL`: https://github.com/slaclab/pytmc/blob/7290ba67a91fca2970a4311cf3a165ab7c5cc696/pytmc/record.py#L637

Since `BOOL` is the only TwinCAT type that uses `BinaryRecordPackage` and because `BOOL` shows up as `TRUE` or `FALSE` while online in TwinCAT, the default should be changed to `TRUE` and `FALSE`, since the usage is overwhelmingly as a boolean flag rather than as a number. In fact, in our projects there isn't a single case where the default values are used intentionally.

If there is some disagreement here then I will soon go through all of my TwinCAT code and copy/paste the same field lines to all of my `BOOL` pragmas.

klauer, 2020-03-24:
The logical terms "TRUE" and "FALSE" may mean something to the PLC programmer which is not obvious to the end-user. These should be handled with care and not left up to interpretation.

ZLLentz, 2020-03-24:
I strongly disagree, this is a minority of cases. In these cases they can be handled with care and customized, like with the power "On" vs "Off" booleans. The strong majority of cases are stuff like enable, reset, execute, done, and busy bits where the end user is getting very confused by "Zero" vs "One" in the typhos screens.

ZLLentz, 2020-03-24:
If we do buy the argument that these sorts of things should be set with care, we have to make things like `ONAM`  and `ZNAM` mandatory includes in the pragma with an exception raised if missing. This would be preferable to letting a PV through that needs to be revised later because it has been given a meaningless default value. I'm currently sitting in a position where I have to track down all these PVs and I would have strongly preferred either a sensible default or a "hey, fix these things now, ok?" message.

klauer, 2020-03-24:
PRs are welcome.

ZLLentz, 2020-03-24:
I made 2 PRs for my two suggestions. There isn't much to implement here, it's just discussing what we want to do...

PR #191 (2020-03-24): MNT: switch default value to most common case
closes #190 
alternative to #192
closed 2020-03-24 by klauer