pcdshub/pytmc #195: Macros in st.cmd

slacAWallace, 2020-04-01:
I can't figure out how to define macros in the makefile to be added to the dbloadrecords call for the pytmc generated db.

It would be nice if ads-deploy added the macros it discovers to the makefile to be defined by the user at some point. For now I am editing the st.cmd to include the macros.

klauer, 2020-04-01:
These should be specified in your IOC `Makefile`. These are the defaults:
https://github.com/pcdshub/ads-ioc/blob/16279f2800ce6ff1cf808dac9fd5e49e974cca70/iocBoot/templates/Makefile.base#L32

In your `st.cmd`, e.g.,
```
DB_PARAMETERS = 'PREFIX=$(PREFIX):,IOCNAME=$$(IOCNAME)',YOUR_MACRO_HERE=YOUR_VALUE_HERE
```

All makefile variables would be excessive. This should be something you configure manually, if necessary.

I'd transfer this to ads-ioc, but it'd be going from slaclab->pcdshub, which I don't think we can do.

slacAWallace, 2020-04-01:
I think the problem is I edit the st.cmd (which for some reason is generated without write permissions), add the PREFIX macro (which doesn't get added to the line for dbloadRecords), and on the next build, the st.cmd is overwritten. I suppose I could disable the st.cmd generation by commenting in the Makefile?

slacAWallace, 2020-04-01:
Makefile
`
IOC_TOP = /reg/g/pcds/epics/ioc/common/ads-ioc/v0.1.5
IOC_INSTANCE_PATH := $(shell pwd)

PROJECT_NAME = Arbiter
PROJECT_PATH := ../../Arbiter/Arbiter.tsproj
PLC = ArbiterPLC

PYTMC_OPTS =
PREFIX = PMPS:LFE:ARBITER

include $(IOC_TOP)/iocBoot/templates/Makefile.base
`

Generated st.cmd
`cd "$(IOC_TOP)"
dbLoadRecords("ArbiterPLC.db", "PORT=ASYN_PLC,")

cd "$(IOC_TOP)"

dbLoadRecords("db/iocSoft.db", "IOC=PMPS:LFE:ARBITER")
dbLoadRecords("db/save_restoreStatus.db", "P=PMPS:LFE:ARBITER:")
`

klauer, 2020-04-01:
There was a clue in there about write permissions: you _should not_ be editing `st.cmd`. You should be editing the makefile and generating the `st.cmd`.

slacAWallace, 2020-04-01:
This makes sense. I can understand why st.cmd is not writeable (although it can be a bit annoying, perhaps just a big ###### AUTOGENERATED DO NOT EDIT ############## at the top?).

Given my Makefile and generated st.cmd, does everything look correct? It seems PREFIX is missing from the ArbiterPLC.db line.

klauer, 2020-04-01:
Your generated results do not look correct as you are 5 months behind on ads-ioc:

`IOC_TOP = /reg/g/pcds/epics/ioc/common/ads-ioc/v0.1.5`

`* 514a42e - Tue, 29 Oct 2019 09:34:59 -0700 (5 months ago) (tag: v0.1.5)`

Please retry with the newest version.