pcdshub/pytmc #266: Functionality for disabled NC axes and integration with ads-ioc

klauer, 2021-05-27:
Question: Should pytmc/ads-deploy _not_ generate a motor record for disabled NC axes?

Sub-question: can a disabled-at-build-time NC axis be dynamically enabled during runtime?

In my opinion:
* If the answer to the sub-question is "no", then pytmc/ads-deploy should skip the axis entirely.
* If the answer to the sub-question is "yes", then pytmc/ads-deploy _may_ (should?) create a motor record

cc @slactjohnson  @ZLLentz @slacAWallace

ZLLentz, 2021-06-01:
I don't think any item in the TwinCAT tree can be dynamically enabled during runtime. I've always needed to reactivate config, so I assumed that a disabled item wasn't even sent to the PLC- could be wrong on the implementation detail, but I'm pretty sure that you cannot (and am 100% sure that you should not) enable NC items dynamically during runtime.

klauer, 2021-06-01:
Then that solves it: pytmc/ads-deploy should skip the axis entirely.