pcdshub/pytmc #282: pytmc link settings should be "CP" instead of "CPP"

klauer, 2022-05-20:
Code:
https://github.com/pcdshub/pytmc/blob/bd2529195298e5b1dbfd96772a7702b2199f4f4b/pytmc/record.py#L496

Otherwise, we will be stuck at the 2Hz updates in the record:

> If the input link specifies CA, CP, or CPP, regardless of the location of the process variable being referenced, it will be forced to be a Channel Access link. This is helpful for separating process chains that are not tightly related. If the input link specifies CP, it also causes the record containing the input link to process whenever a monitor is posted, no matter what the record’s SCAN field specifies. If the input link specifies CPP, it causes the record to be processed if and only if the record with the CPP link has a SCAN field set to Passive. In other words, CP and CPP cause the record containing the link to be processed with the process variable that they reference changes.

ZLLentz, 2022-05-20:
So, CPP doesn't cause the target PV to process in the case where it has some defined schedule, i.e. a scan or some io intr, and this change would make it always process right when we hit the target PV?

klauer, 2022-05-23:
I added some background information to the PR text that should hopefully clarify this a bit more. I think you have the right idea, though.

PR #283 (2022-05-23): FIX: CP link instead of CPP link
## Context

Closes #282 

Prior to this PR, we are erroneously limited to 2Hz updates for all IOC->PLC PV links due to the 0.5 second SCAN rate configured for linked records. What we want is the IOC to write to the PLC on *every* update, and failing that at 2Hz.

## Background

* A `CPP` link means "channel access link which causes processing (when the monitor is fired), **only if** the record's scan field is set to Process Passive
* A `CP` link means "channel access link which causes processing (when the monitor is fired), regardless of the record's SCAN field [... cut]
closed 2022-05-23 by klauer