pcdshub/pytmc #313: lfe and kfe vac IOCs do not build

ZLLentz, 2023-06-20:
Error is duplicate records

Need to investigate to see if there's a pytmc bug or if we somehow have production PLCs whose IOCs are broken

To reproduce:
- clone the repo
- update ads-ioc to latest
- `make` with `pytmc` at master

ZLLentz, 2023-06-20:
After investigation, this might become an issue in those repos instead of this one

klauer, 2023-06-20:
Duplicate records is typically a library versioning misconfiguration

https://confluence.slac.stanford.edu/display/PCDS/Beckhoff+Flight+Rules#BeckhoffFlightRules-pytmcreports%22duplicaterecords%22

Which records is it?

ZLLentz, 2023-06-20:
It's general lib, so maybe it's just standard library misconfiguration + never built with modern ads-ioc/pytmc with duplicate checking?

klauer, 2023-06-20:
That must be it (🤞 )

ZLLentz, 2023-06-21:
lfe vac plc actually does build, I must have gotten confused
kfe vac is definitely the general lib thing and has the problem with latest tag as well as on master
issue created in kfe vac repo

closed 2023-06-21 by ZLLentz